bullestock/4xidraw #3: Limited by physical range of motion error

Prithul0218, 2019-08-18:
My large, detailed prints often stops midway, and the Inkscape extensions says: "Warning: 4xiDraw movement was limited by its physical range of motion. If everything looks right, your document may have an error with its units or scaling. Contact technical support for help!"

However, I set my software limits to 5000 mm for all axis and I am printing an A4 size picture. So there is no way it ran into software limit, the warning is a false trigger and now my 2 hour drawing is incomplete. The pen did not even return to home,

It is actually happens with crosshatch style drawings, saved as a SVG file.  I don't get this problem on the example drawings. 

Please take a look.

bullestock, 2019-08-18:
@Prithul0218 I think you forgot to attach the file you are having problems with.

bullestock, 2019-11-08:
No response for over two months.

closed 2019-11-08 by bullestock