bullestock/4xidraw #4: Servo only one postition even low or raise. Can't raise during printing (On CNC machine, with sg90 servo).

caogia, 2019-10-22:
hi, I uploaded grbl servo sketch to my uno CNC machine, It worked well
The servo can turn but only one position, doesn't seem to raise up.
my servo is china sg90 toy servo, it work well with arduino servo sample sketch but can't work properly on axidraw.
Please help.

madc, 2019-11-08:
I had to switch the values for `Pen UP` and `Pen Down` in `4xiDraw Control` > `Setup`. Hope that helps..

[image]

bullestock, 2019-11-08:
> I had to switch the values for `Pen UP` and `Pen Down` in `4xiDraw Control` > `Setup`. Hope that helps..

That probably means that your servo is rotated 180 degrees with respect to mine.

closed 2019-11-08 by bullestock