bullestock/4xidraw #8: Unexpected response from GRBL

MCTaylor17, 2020-02-04:
I'm just doing my first few test prints and I'm getting the same error at the end of each print:
```
Error: Unexpected response from GRBL.
   Command: G90
   Response: 
Error: Unexpected response from GRBL.
   Command: M3 S90
   Response: Grbl 0.9i ['$' for help]
```
It's a little strange since I have the pen set to 45 degrees so it shouldn't even be giving the `M3 S90`.

bullestock, 2020-02-04:
Since you get the version response from GRBL, could it be that your
controller is resetting?

misan, 2020-02-12:
If the servo is powered from the Arduino's 5V it could trigger a brownout reset if it consumes more power than what the onboard regulator.

bullestock, 2020-02-12:
> If the servo is powered from the Arduino's 5V it could trigger a brownout reset if it consumes more power than what the onboard regulator.
Good point, that could very well be what's happening.

misan, 2020-02-13:
@MCTaylor17 Can you try sourcing the +5V from another regulator (for example a 7805)? (Remember the GND of all supplies has to be connected together). Or just run the same code with the servo disconnected to see if the problem goes away.

MCTaylor17, 2020-02-13:
I just tried with a fresh install of GRBL (the one linked on instructibles) with only the USB plugged in and got the same error.

Before troubleshooting this further, I am curious about upgrading to GRBL v1.1.  Is there a fork that implements the servo commands that I could test?

misan, 2020-02-14:
@MCTaylor17 So the servo was disconnected, right? 

Are you using any end-stops?

As Torsten mentioned above, something is causing Arduino to restart. 
If the cause is not the servo, it has to be something else. 
GRBL version is not the problem and you would need to apply the changes to keep it supporting the servo which is not a native feature of GRBL and thus not part of the upgrade process.

pdiddyparkour, 2020-04-30:
I also encountered the same issue with my setup. What I had to do was disable the Arduino Uno's automatic reset by using a capacitor in between the reset and ground pin. See the following instructions: https://forum.arduino.cc/index.php?topic=363128.0
I used a 1000uF capacitor I had laying around (probably way overkill) and it fixed the issue for me - no more pop up error box! Big thanks to the folks on the Arduino forums for figuring this out. I also encountered timeout errors with long/slow pen arm movements. That issue went away when I set the GRBL accelerations & movement speeds to much higher values via Universal Gcode Sender.

MCTaylor17, 2020-05-01:
I'm sorry I haven't had time to look at this for a while.  After the pandemic started, I shifted my attention to PPE design and production: https://www.thingiverse.com/ADHDroid/designs

I'm currently rapid prototyping a new ear saver design that should make donning/doffing face masks much much easier.

@pdiddyparkour I think I actually saw the post you're referring to but it didn't seem to do the fix.  Unless I'm mixing things up with a different problem, I remember using the exact capacitor they suggested.  Maybe going overkill is the solution...

I have way too many projects on the go so I'll call this one closed and consider revising the issue in the future if I ever get back around to it.

closed 2020-05-01 by MCTaylor17