bullestock/4xidraw #9: Inaccurate distances in Inkscape

mahtDFR, 2020-02-17:
I have max travel of 300x, 300y. In universal Gcode sender this works as expected, however the Inkscape extension always runs past the end of travel when using a4 pages. It seems its producing drawings 2x the actual size, so A4 in Inkscape becomes A2 when plotting.

Which setting should I adjust to fix this?

bullestock, 2020-02-18:
You can edit the file fourxidraw_conf.py and change the setting DPI_16X.

However, it sounds as if your stepper drivers are set to 8 x microstepping and not 16 x. Changing that should also fix the issue.

mahtDFR, 2020-05-10:
My controller blew out and I only just replaced it, so I just got around to trying this out.

I'm using an UNO CNC shield and [according to this](https://courses.ideate.cmu.edu/16-376/s2019/ref/text/hardware/cnc-shield.html#microstepping-setup) I'm sure it's set to 16x microstepping - M0, M1, and M2 are pulled up.

I have reconfigured my x and y steps/mm in GRBL as follows:

``$100=80.071 (x, step/mm)``
``$101=80.071 (y, step/mm)``

I have edited fourxidraw_conf.py as follows:

``DPI_16X = 80.07064		# DPI ("dots per inch") @ 16X microstepping. Standard value: 100 steps per mm.``

I've actually tried a few values here and there doesn't seem to be any effect when calling it from Inkscape.

There seems to be some strange offsets and sizing issues still. The output is larger on the page than the input. Any advice would be appreciated!

steersteer, 2020-05-21:
Having this problem also. If a put a small image in the bottom corner of the page it draws it twice as big in the centre of the page