bullestock/4xidraw #10: Plotter won't draw more than 300mm on x-axis and 210mm on y-axis

gello0, 2020-04-08:
With other gcode generators I can do it but it after reaching 210mm or 300mm one of the axis wont move 
heres my settings 
```$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 0    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 0    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 80.000    (x, step/mm)
$101 = 80.000    (y, step/mm)
$102 = 250.000    (z, step/mm)
$110 = 15000.000    (x max rate, mm/min)
$111 = 15000.000    (y max rate, mm/min)
$112 = 0.000    (z max rate, mm/min)
$120 = 1200.000    (x accel, mm/sec^2)
$121 = 1200.000    (y accel, mm/sec^2)
$122 = 1200.000    (z accel, mm/sec^2)
$130 = 10000.000    (x max travel, mm)
$131 = 10000.000    (y max travel, mm)
$132 = 10000.000    (z max travel, mm)```

bullestock, 2020-04-08:
What is your document size set to in Inkscape?

gello0, 2020-04-08:
Inkscape document properties are set to 
``` width 190```
 ```height 370```

bullestock, 2020-04-08:
And if you set it to e.g. 500 x 500 mm?

gello0, 2020-04-09:
Same behavior with 500x500mm, also noticed `Auto-rotate page when printing` option wont work with this page size. 

[500x500_test.svg.zip heres my svg file](https://github.com/bullestock/4xidraw/files/4454439/500x500_test.svg.zip)

[Video of plotter](https://drive.google.com/file/d/1-FBYmlUCGsXrNZ6kIstc-FUZxbYdUQdz/view?usp=sharing)

I also tried [this simple gcode](https://github.com/bullestock/4xidraw/files/4454453/output_0028.gcode.txt) to check if $130 and $131 are causing it to stop at 300mm but it travels to x-axis and y-axis travel to 370mm.

BerndNeuhauser, 2021-04-24:
i build a larger Version of the 4xiDraw. And also got the Message:
"Warning: 4xiDraw movement was limited by its physical range of motion.
If everything looks right, your document may have an error with its
units or scaling. Contact technical support for help!"

Done:
I changed $130 and $131 to get bigger physical range. Doesnt help.
i changed the Document properties and SVG Scaling in Inkscape. Doesnt help.
i searched for a solution but only found open issus for it.

So i reverse engeneered the code to see wher is a limitation of the size.

Solution:
1. Go to your inkscape directory
2. open fourxidraw_conf.py
3. change

PageWidthIn
PageHeightIn

to your physical size.

Even if the comment in the file says "The values listed here are ignored when called via Inkscape." !!!!IT´S A LIE!!!!! It matters!

If i helped someone, please say "thank you" :D

bullestock, 2021-04-24:
@BerndNeuhauser thanks, I've added a section to the README about this.

I think if you run the script with the --logSerial option, it should output the G-code.