bullestock/4xidraw #13: Python 3 Support

MatthewJohnMorris, 2020-12-23:
Hi, and thanks for providing such a useful extension!

I have managed to get it working with Python 3 & modern PySerial - would you be interested in a patch request for this? The main difference lies in the difference between P2 and P3 on string handling.

Thanks,
Matt

bullestock, 2020-12-25:
Sure, shoot me a pull request and I'll be right on it.

MatthewJohnMorris, 2020-12-26:
PR is now in at https://github.com/bullestock/4xidraw/pull/14 - let me know if there's any questions. I've managed to fix all the deprecation warnings on the test svgs I've been printing, so it now runs cleanly without popping up any messages at the end.

MatthewJohnMorris, 2020-12-28:
I see you merged https://github.com/bullestock/4xidraw/pull/14 - just letting you know I'm going to have another PR soon - I'm trying to work my way through as many examples as I can to hit code paths not exercised by AxiDraw_First.svg, now my 4xidraw build quality is improved enough to run more complex stuff.

One example at "examples/test and cal/SmalltestPattern.svg" has thrown up 2 more deprecation warnings and a change in behaviour that crashes when running Inkscape 1.x so I'll put a PR in once I've got that SVG working properly.

MatthewJohnMorris, 2020-12-28:
I've now got SmalltestPattern.svg working so here's a PR for the changes required. It includes an improvement to Python importing that should better separate out the Python 2 and Python 3 code paths from impacting each other. PR is at https://github.com/bullestock/4xidraw/pull/15

bullestock, 2020-12-28:
Thanks a lot for the patches, haven't used my own 4xiDraw for a long time (there was no MiniMakerFaire this year due to COVID, otherwise I would probably have used it there).

MatthewJohnMorris, 2020-12-28:
Happy to help out, I currently do a reasonable amount of Python coding in the day job so I'm as well placed to do the patches as anyone. I will probably upgrade to a full-on Axidraw at some point - but I figure I should pay my dues first and figure out how the engineering tradeoffs go, rather than just splashing out the money!

bullestock, 2020-12-29:
Note that the AxiDraw (AFAIK) does not use G-gode, so the 4xidraw plugin is
not compatible with it.

MatthewJohnMorris, 2020-12-29:
Interesting - I didn't know that... yes they have their own protocol as per https://evil-mad.github.io/EggBot/ebb.html#syntax - wonder why... I guess it helps them optimise for their hardware?

misan, 2020-12-29:
@MatthewJohnMorris It is my understanding EBB moves are only at a constant speed, thus any acceleration has to be simulated by chaining moves with increasing speeds, which requires multiple commands for a single straight-line move if accel/decel is needed. 

I reckon they use EBB as it was used for previous projects.

MatthewJohnMorris, 2020-12-31:
Interesting, I will have a look at that at some point...

@bullestock I've raised snother PR https://github.com/bullestock/4xidraw/pull/16 - this is to take care of a deprecation warning and clear up handling of pixel extents. I seem to be able to run pretty much anything else now.

One question - I see the "distance travelled" for pen up/down shown when "reportTime=True" isn't actually hooked up to any calcualtion at the moment, so always returns zero. Would there be any interest in my adding support for that?

bullestock, 2020-12-31:
> One question - I see the "distance travelled" for pen up/down shown when "reportTime=True" isn't actually hooked up to any calcualtion at the moment, so always returns zero. Would there be any interest in my adding support for that?

It's not something that I have wished for personally, but if you're bored, go ahead :-)

MatthewJohnMorris, 2020-12-31:
> It's not something that I have wished for personally, but if you're bored, go ahead :-)

I'll see how bored I get with the family jigsaw puzzles at home :)

MatthewJohnMorris, 2020-12-31:
I think anything else I work on at this point will be an enhancement rather than Python3 stuff, so may as well close the issue

closed 2020-12-31 by MatthewJohnMorris