bullestock/4xidraw #17: Why is the option Greyed out in Inkscape 1.0.1?

saintofinternet, 2021-01-01:
Why is the option Greyed out in Inkscape 1.0.1?

bullestock, 2021-01-01:
Which version of the plugin are you using?

saintofinternet, 2021-01-03:
the one from Github page....

now the option has started to work as i copied the .py and .inx files in the "Extension" folder as well.

but there is a issue with the Python version as now Python 2 Syntax also is giving an error I suppose

how can this all be fixed to work with the Inkscape 1.0.1??

[image]

MatthewJohnMorris, 2021-01-24:
The current code should work In Inkscape 1.0.1/Python 3 - I'm using it myself. But I'm not sure exactly what has been copied across - [the current version of "4xidraw.inx"](https://github.com/bullestock/4xidraw/blob/b1a8d3a42ce7d57cfd57785f665be9f158bf67fe/inkscape%20driver/4xidraw.inx#L3) should give you a menu option "4xiDraw" with submenu called "4xiDraw Control..." and a grayed out one called "Hatch fill...". You seem to be getting a menu option called "4xiDraw Tools", with other submenu options, not quite sure how?