bullestock/4xidraw #18: Showing error when using in Inkscape 1.0.1

AkhilGKrishnan, 2021-01-02:
Traceback (most recent call last):
  File "fourxidraw.py", line 1416, in <module>
    e.run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 140, in run
    self.save_raw(self.effect())
  File "fourxidraw.py", line 310, in effect
    self.serialPort = grbl_serial.openPort(self.options.logSerial)
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/grbl_serial.py", line 90, in openPort
    foundPort = findPort()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/grbl_serial.py", line 14, in findPort
    from serial.tools.list_ports import comports
  File "/Applications/Inkscape.app/Contents/Resources/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Applications/Inkscape.app/Contents/Resources/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Applications/Inkscape.app/Contents/Resources/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

bullestock, 2021-01-03:
You forgot to mention what OS you are using.

AkhilGKrishnan, 2021-01-10:
I'm using MacOS BigSur

MatthewJohnMorris, 2021-01-24:
TL;DR: it appears this may be an issue with Python 3.8.5 (Inkscape 1.0.1's default version) and MacOS Big Sur. Try 3.8.7 and see if that improves things (instructions below).

The version of Python in Inkscape 1.0.1 appears to be 3.8.5 (at least, that's what I get when I run "python --v" on the copy on my Windows box at C:\Program Files\Inkscape\bin). From https://www.python.org/downloads/mac-osx/ I see that Python 3.8.5 was released on 20Jul2020. However a search for "dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found" led me to this Python issue https://bugs.python.org/issue41100 - from which it appears that problems with MacOS Big Sur surfaced on 24Jul2020. So it looks like Python 3.8.5 is broken on MacOS Big Sur.

The problem may be fixed in 3.8.6 (released 24Sep2020) or 3.8.7 (released 21Dec2020). What I recommend is that you grab Python 3.8.7, and follow the instructions at https://wiki.inkscape.org/wiki/index.php/Extension_Interpreters to point Inkscape to use that version for the 4xidraw extension. Hopefully this may resolve the issue.

Also, it's possible that just-released update Inkscape 1.0.2 will have an updated version of Python, but no guarantee there. You can try it from here: https://inkscape.org/release/inkscape-1.0.2/